SuperFlyTV/xkeys #2: Do you have time to dev this more?

Planks, 2018-01-21:
Hey man

I have an XK-68 Jog+Shuttle that I'd really like to get working with a nodejs app that I use. I've installed your test script, and added a config line for my device manually, but the settings have been somewhat guesswork

Using the following lines, I'm able to read button presses from the device fine, and the button/joystick signals from the jogwheel, but the shuttle only reads 7 steps to the right and not to the left.

```
		XK68JOG: {	// This has not been tested
		identifier: 'XK-68 Jog-Shuttle',
		productId: [1114],
		columns: 	10,
		rows: 		8,
		hasPS: 		true,
		hasJoystick: 1,
		bankSize: 	32,
	},
```

This could be me being hacky and guessing parameters, or just that you've never had a chance to develop and bugfix that side because you don't own a device with the jogwheel.

Is this library something you have time to revisit and work with me to get working?

nytamin, 2018-01-21:
Sure, we could figure out how it works together!
I didn't have that Xkeys available so the implementation was a pure guess to be honest, but if we log the output from the device, it should be fairly easy to implement the support correctly.
I could get back to you tomorrow with a test-script to run, sounds good?

Planks, 2018-01-21:
Perfect! I will have time at some point or another to run through whatever needs done

nytamin, 2018-01-22:
Here's a quick script I put together just now:
[logHMI.zip](https://github.com/SuperFlyTV/xkeys/files/1650937/logHMI.zip)
It simply logs all data to a file (log.txt).
Connect your xkeys, run the script, pull the shuttle and then post your results here!

It would also be very helpful it you edit the logfile after recording, adding explanations on what you where doing, like this:
Log.txt:
2018-1-22 07:18:26.310 Listening to device (1):
2018-1-22 07:18:26.312 manufacturer   P. I. Engineering
2018-1-22 07:18:26.315 product        XK-24 HID
2018-1-22 07:18:26.320 vendorId       1523
2018-1-22 07:18:26.324 productId      1029
2018-1-22 07:18:26.327 usage          1
**Pressed the button on top left:**
2018-1-22 07:18:27.228 1 data 01000100000000095f5a00000000000000000000000000000000000000000000
**Released the button on top left:**
2018-1-22 07:18:27.411 1 data 0100000000000009600c00000000000000000000000000000000000000000000
**Continued clicking some buttons:**
2018-1-22 07:18:27.603 1 data 010002000000000960c800000000000000000000000000000000000000000000
2018-1-22 07:18:27.739 1 data 0100000000000009615000000000000000000000000000000000000000000000
2018-1-22 07:18:27.918 1 data 0100040000000009620400000000000000000000000000000000000000000000
2018-1-22 07:18:28.056 1 data 0100000000000009628d00000000000000000000000000000000000000000000
2018-1-22 07:18:28.249 1 data 0100080000000009635400000000000000000000000000000000000000000000
2018-1-22 07:18:28.368 1 data 010000000000000963c300000000000000000000000000000000000000000000
2018-1-22 07:18:28.594 1 data 010010000000000964a300000000000000000000000000000000000000000000
2018-1-22 07:18:28.688 1 data 0100000000000009650500000000000000000000000000000000000000000000
2018-1-22 07:18:28.856 1 data 010020000000000965ac00000000000000000000000000000000000000000000
2018-1-22 07:18:28.952 1 data 0100000000000009660900000000000000000000000000000000000000000000

Planks, 2018-01-22:
Please find attached,

I did a bunch of these as separate logs so as to not forget where I was at in commenting ;)

[jog full clockwise.txt](https://github.com/SuperFlyTV/xkeys/files/1651958/jog.full.clockwise.txt)
[jog full counterclockwise.txt](https://github.com/SuperFlyTV/xkeys/files/1651959/jog.full.counterclockwise.txt)
[shuttle left then centre.txt](https://github.com/SuperFlyTV/xkeys/files/1651960/shuttle.left.then.centre.txt)
[shuttle right then centre.txt](https://github.com/SuperFlyTV/xkeys/files/1651961/shuttle.right.then.centre.txt)
[soft key rows.txt](https://github.com/SuperFlyTV/xkeys/files/1651962/soft.key.rows.txt)
[button on the top.txt](https://github.com/SuperFlyTV/xkeys/files/1651963/button.on.the.top.txt)

nytamin, 2018-01-22:
Perfect!
I've pushed some updates to the dev-branch.
If you could pull that and test if it works please?

Fingers crossed!

Planks, 2018-01-22:
Hiya

Seems to work great

I had to add these lines to your sample javascript in the readme that gives a human readable output 

`myXkeysPanel.on('shuttle', deltaPos => {
	console.log('Shuttle position has changed: '+deltaPos);
});`
[test.txt](https://github.com/SuperFlyTV/xkeys/files/1653477/test.txt)

Thanks very much for your help!

nytamin, 2018-01-22:
Great! I'll push the update to npm shortly.

Planks, 2018-01-22:
Small comment - if I hold the shuttle in a non-neutral position, and press another key, an extra shuttle change of state (eg. from 3 to 3) is logged in the console even though no change has taken place

`Shuttle position has changed: -1
Shuttle position has changed: -2
Shuttle position has changed: -3
Shuttle position has changed: -4
Shuttle position has changed: -5
Key pressed: 69
Shuttle position has changed: -5
Key released: 69
Shuttle position has changed: -5
Shuttle position has changed: -4
Shuttle position has changed: -3
Shuttle position has changed: -2
Shuttle position has changed: -1
Shuttle position has changed: 0
`

nytamin, 2018-01-23:
Thanks, I've pushed a minor update to dev, could you verify that it works?

Planks, 2018-01-25:
I’m away from home until Sunday, I’ll check then!

nytamin, 2018-01-31:
ping @Planks, if you have time to check it? :)

Planks, 2018-01-31:
Checked now, works, thanks for the reminder.

You should add the shuttle code to your demo code in the readme!

nytamin, 2018-01-31:
good idea! I'll push this to the master branch and to a new npm version right away!
Thanks for the help!

closed 2018-01-31 by nytamin