SuperFlyTV/xkeys #4: `usage` has been removed in node-hid

engwan, 2018-04-25:
Refer to: https://github.com/node-hid/node-hid/issues/120

This library depends on `usage` to find the right HID device.

I made it working by using the `interface` field instead.

Device with `interface` of `0` is the correct one to listen to.

nytamin, 2018-04-25:
Good catch! Please submit a PR and I'll put it in.
We should probably update the dependency of node-hid-version as well, while we're at it

engwan, 2018-04-26:
Would you know how it works on Windows?

Are `usage` and `usagePage` removed in Windows too? Is `interface` also available on Windows?

nytamin, 2018-04-26:
I see that usage and usagePage are available on windows, and interface too. So it should be okay switching to using the interface attribute instead

nytamin, 2020-10-27:
This has been addressed in https://github.com/SuperFlyTV/xkeys/commit/2883c466f2ea26585a14b6e9765fa4146ba17554 , released in versions `>=1.0.0` Better late than never I suppose...

closed 2020-10-27 by nytamin