SuperFlyTV/xkeys #5: Xkeys Switches

CodyBranch, 2018-06-26:
I was looking for a way to use Xkeys Switches with node, http://xkeys.com/XkeysSwitches/index.php

I wanted to see if anyone had looked into this at all before I jumped in. I use this code to interface with the XK-12 with Jog and it's pretty slick and simple to use.

nytamin, 2018-06-27:
I personally haven't used any of those switches, but if they work similarly as the panels it should be easy to implement.
Please submit a PR when you've got it working! :-)

CodyBranch, 2018-06-27:
I got two of the three buttons working on the XK-2512-BU (http://xkeys.com/accessories/TriggerSwitch.php)

Used the XK12 USB switch interface. (http://xkeys.com/xkeys/xkswi.php)

Seemed pretty easy, I just added the below code to the index.js file. I'll do some further testing/investigating to see why the top button isn't working. The guys from Pi Engineering said their software can only control 2 of the 3 buttons too, so who knows.

I just kind of tried stuff with the columns, rows and banks options. I had 6 of the switches plugged in and could use 2 buttons on all 6 of them.

`	XK12: {	// This has not been tested
		identifier: "PI4 Switch Interface",
		productId: [1195],
		columns: 	12,
		rows: 		12,
		hasPS: 		false, // unknown
		bankSize: 	80, // unknown
		banks: 		6
	},`

nytamin, 2018-06-27:
Cool!
You can monitor the raw data by running logHMI.js, that might give you a clue if the third button is sent over the wire at all.

CodyBranch, 2018-06-27:
It doesn't look like the 3rd button does anything. Which would make sense since its works with 12 switches and 6 devices x 2 buttons = 12 switches.

I have a 3 switch USB connector coming this week to see if all 3 buttons will work with it. If so I'll report back with what I find, then we can add this in.

Can also confirm this package works with a XK-12JOG as I've been using it pretty extensively the last 2 weeks and it's been solid.

nytamin, 2018-06-27:
Great, I'll update the readme, thanks!

closed 2020-10-27 by nytamin