SuperFlyTV/xkeys #8: setBacklightIntensity always disables red back-lights

jonwyett, 2019-06-05:
setBackLightIntensity checks for a 'banks' property of this.deviceType that is not defined in the [PRODUCTS] object. Since the property doesn't exists it always sets the red bank to 0 intensity.

Simple solution, add the banks property to the [PRODUCT] object:
```
XK24: {
		identifier: "XK-24",
		productId: [1029,1028,1027,1249],
		columns: 	4,
		rows: 		6,
		hasPS: 		true,
		bankSize: 	32,
		banks:		2 //added the banks property
	}
``` 
A more complex solution would be to add a red bank parameter to the function:
```
setBacklightIntensity(blueIntensity, redIntensity) {
    	blueIntensity = Math.max(Math.min(blueIntensity,255),0);
	redIntensity = Math.max(Math.min(redIntensity,255),0);
    	var message = [];
    	if (this.deviceType.banks === 2 ) {
    		message = this.padMessage([0,187,blueIntensity, redIntensity]); //added redIntensity
    	} else {
    		message = this.padMessage([0,187,blueIntensity]);
    	}

    	this.write(message);
	}
```

Jheronymus, 2020-10-17:
nice catch!

nytamin, 2020-10-27:
This has been addressed in https://github.com/SuperFlyTV/xkeys/commit/a75d330f2161cf8b9d191feec2985ff14a36689d , released in versions `>=1.0.0`.

closed 2020-10-27 by nytamin