SuperFlyTV/xkeys #9: Log.txt for XK-4 HDI keys

KnutHelstad, 2020-10-27:
2020-10-15 09:10:32.989 Listening to device (1):
2020-10-15 09:10:32.999 manufacturer   P. I. Engineering
2020-10-15 09:10:32.999 product        XK-16 HID
2020-10-15 09:10:33.003 vendorId       1523
2020-10-15 09:10:33.004 productId      1049
2020-10-15 09:10:33.016 usage          1
npm run

Works but gets index 0,8,16,32 for buttons but diode index is 0,1,2,3

nytamin, 2020-10-27:
Hello! Thanks for taking the time to help out.
Could you run the logHMI.js script once again, and be sure to also click the buttons on the xkeys, in order.
It would also help if you could note down which buttons, and in which order they where pressed.
An example for how it looks like for an xk-24 like this:
```
Listening to device (1):
manufacturer   P. I. Engineering
product        XK-24 HID
vendorId       1523
productId      1029
usage          1
// I pressed button 1 here
1 data <Buffer 01 00 01 00 00 00 00 07 30 cf 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
1 data <Buffer 01 00 00 00 00 00 00 07 31 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
// I pressed button 2 here
1 data <Buffer 01 00 02 00 00 00 00 07 32 4e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
1 data <Buffer 01 00 00 00 00 00 00 07 32 a3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
// I pressed button 3 here
1 data <Buffer 01 00 04 00 00 00 00 07 33 a6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
1 data <Buffer 01 00 00 00 00 00 00 07 33 f7 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
// I pressed button 4 here
1 data <Buffer 01 00 08 00 00 00 00 07 34 be 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
1 data <Buffer 01 00 00 00 00 00 00 07 35 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
// I pressed button 5 here
1 data <Buffer 01 00 10 00 00 00 00 07 36 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
1 data <Buffer 01 00 00 00 00 00 00 07 36 4f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>

```

KnutHelstad, 2020-10-27:
npm run2020-10-27 09:01:44.042 Listening to device (1):
2020-10-27 09:01:44.043 manufacturer   P. I. Engineering
2020-10-27 09:01:44.047 product        XK-16 HID
2020-10-27 09:01:44.048 vendorId       1523
2020-10-27 09:01:44.048 productId      1049
2020-10-27 09:01:44.049 usage          1
2020-10-27 09:01:46.703 1 data 00000100000003b30af800000000000000000000000000000000000000000000
2020-10-27 09:01:47.004 1 data 00000000000003b30c2500000000000000000000000000000000000000000000
2020-10-27 09:01:48.084 1 data 00000001000003b3105400000000000000000000000000000000000000000000
2020-10-27 09:01:48.396 1 data 00000000000003b3118f00000000000000000000000000000000000000000000
2020-10-27 09:01:49.723 1 data 00000000010003b316b600000000000000000000000000000000000000000000
2020-10-27 09:01:50.044 1 data 00000000000003b317f100000000000000000000000000000000000000000000
2020-10-27 09:01:50.830 1 data 00000000000103b31aff00000000000000000000000000000000000000000000
2020-10-27 09:01:51.100 1 data 00000000000003b31c0d00000000000000000000000000000000000000000000

KnutHelstad, 2020-10-27:
With som comment

npm run2020-10-27 09:01:44.042 Listening to device (1):
2020-10-27 09:01:44.043 manufacturer   P. I. Engineering
2020-10-27 09:01:44.047 product        XK-16 HID
2020-10-27 09:01:44.048 vendorId       1523
2020-10-27 09:01:44.048 productId      1049
2020-10-27 09:01:44.049 usage          1
// I pressed button 1 here
2020-10-27 09:01:46.703 1 data 00000100000003b30af800000000000000000000000000000000000000000000
2020-10-27 09:01:47.004 1 data 00000000000003b30c2500000000000000000000000000000000000000000000
// I pressed button 2 here
2020-10-27 09:01:48.084 1 data 00000001000003b3105400000000000000000000000000000000000000000000
2020-10-27 09:01:48.396 1 data 00000000000003b3118f00000000000000000000000000000000000000000000
// I pressed button 3 here
2020-10-27 09:01:49.723 1 data 00000000010003b316b600000000000000000000000000000000000000000000
2020-10-27 09:01:50.044 1 data 00000000000003b317f100000000000000000000000000000000000000000000
// I pressed button 4 here
2020-10-27 09:01:50.830 1 data 00000000000103b31aff00000000000000000000000000000000000000000000
2020-10-27 09:01:51.100 1 data 00000000000003b31c0d00000000000000000000000000000000000000000000

nytamin, 2020-10-27:
Thanks! Just to verify, it is actually am XK-4, right? Strange that the device reports itself as a XK-16.. oh well, they're probably re-using the same chip for all of the one-row devices.

@KnutHelstad how does the library work with the XK-4 right now? As I'm adding the test, it seems like it might be already working, could you verify?
( please run the `examples/xkeys.js` script and see if you get the evens in the console, as well as the blue light lit up on the button itself when you press it)

KnutHelstad, 2020-10-27:
on v 0.2.1 i get

PS C:\nodeProjects\inlinervision> node xkeytest
// I pressed button 1 here
Key pressed: 0
Key released: 0
// I pressed button 2 here
Key pressed: 8
Key released: 8
// I pressed button 3 here
Key pressed: 16
Key released: 16
// I pressed button 4 here
Key pressed: 24
Key released: 24

The exsact number is XKS-04-USB-R

nytamin, 2020-10-27:
Yeah that seems to be right, the x-keys devices have their numbering like that; 0 being the lop-left, 1 is below it, 8 is to the right of it etc..

The indexes for your keys are actually 0, 8, 16, 24, so `myXkeysPanel.setBacklight(8, true)` should work for you (?).

KnutHelstad, 2020-10-27:
Now.

`myXkeysPanel.setBacklight(Index, true)` works when using index 0,1,2,3

nytamin, 2021-01-06:
closing this, feel free to reopen if there are any lingering issues

closed 2021-01-06 by nytamin