SuperFlyTV/xkeys #24: Support for XK124-Tbar

laurensvane, 2021-01-06:
Good day,

Could you build in support for the XK124-T-bar?
I started with the following snippet in products.js which enabled all the buttons for me:
```
    XK124TBAR: {
        identifier: 'XK-124 T-Bar',
        productId: [1278],
        columns: 16,
        rows: 8,
        hasPS: false,
        banks: 2,
        bankSize: 128,
        disableKeys: [108, 109, 110, 111]
    },
```

If i am correct, support for the t-bar has to be built in seperatly, because it reuses the dataslots from the buttons 108-111?

i ran loghmi.js, i will try to 
attach the exported logfile to this issue.
[log.txt](https://github.com/SuperFlyTV/xkeys/files/5775874/log.txt)

nytamin, 2021-01-06:
Thanks for the log file!
There was a new version (1.1.0) released just a few hours ago which added support for the [XKE-T bar](https://github.com/SuperFlyTV/xkeys/pull/23), did you try that?

laurensvane, 2021-01-06:
I just installed on a new raspberry pi and saw that the tbar is working now.
The issue must have crossed the release.
Maybe an idea to also include the on('tbar') in the example javascript code.

nytamin, 2021-01-06:
Good that it's working! I'll update the readme with an example.

closed 2021-01-06 by nytamin