SuperFlyTV/xkeys #26: Console log statement becomes rather spammy

mountaindude, 2021-01-10:
The console log statement [here](https://github.com/SuperFlyTV/xkeys/blob/8f06cc4681e691a8513c911dbc9be8b43e2e4d0b/src/xkeys.ts#L337) can become rather annoying at times. 

I have an API on top of this NPM lib, with the API offering control of backlights - including intensity - via a single call.
At times the API is called frequently (many times every second) and the console then becomes just a long list of 

```
this.deviceType.banks 2
```

Is there a reason to keep that console.log there, or should I send in a PR to get rid of it?

Love the lib btw - super useful!

nytamin, 2021-01-15:
Thanks for bringing it up!
Yeah it shouldn't be in there, I've fixed it now in version `1.1.1`.

If you have any suggestions or want to add any convenience methods that makes it easier to use, feel free to submit a PR!

closed 2021-01-15 by nytamin

mountaindude, 2021-01-15:
Yeah I might add some convenience functions that expose a bit more of the underlying HID features... Always a balance of keeping things abstracted and giving developers flexibility. 

Funny - I got a X-keys panel some years ago but never got around to using it due to lacking support in mac OS. 
And now I realized the solution was so simple - just use Node to whip up an API server and boom! Good stuff.