SuperFlyTV/xkeys #27: Handling of USB disconnect/reconnect

mountaindude, 2021-01-27:
Been using the library for a while, works great!

...Except when a) you're on mac OS, and at the same time b) the panel is connected far down the USB tree, in a USB hub connected to a USB hub connected to a ... 
Due to some limitation in mac OS APIs the path field will come back empty from node-hid, making the panel unusable. 
There is ongoing work over at node-hid re this, and this issue is not in any way caused by the xkeys lib.

One thing is still an issue with the xkeys lib though: If a panel is disconnected while in use, the lib will loose track of the panel even if it's reconnected a minute (or a second) later.

That's fine really, but it would be nice with a solid process for handling disconnects/reconnects. 

First idea:

1. The app should keep track of the USB path of the panel it uses, as well as keep a variable myXkeysPanel for interacting with the panel in question.
2. Set up USB listeners for add/remove events, maybe using https://github.com/MadLittleMods/node-usb-detection
3. In the "removed" event handler, check if it was the panel in use that was removed. If so set the corresponding myXkeysPanel variable to null. The rest of the app must be smart enough to realize that `null` means "panel not connected". 
4.   In the "add" event handler do the reverse of 3. I.e. determine whether it was the panel-in-use that was reconnected, and if so do a new HID enumeration, get connected panels and eventually something like a `myXkeysPanel = new XKeys(pathToPanel)`

Should work I think. One possible issue is that I'd introduce a new state (and tracking of that state) in the app ("panel available status") which is always a source of headaches.  

Or: Have you achieved this in some other/better way?

Cheers,
Göran

mountaindude, 2021-01-27:
... on the other hand...

node-usb-detection is good at doing exactly that - detecting add/removals of USB devices. 
But it doesn't report back the path or some other unique identifier that is shared with node-hid (and thus xkeys lib). 
Which means it will be difficult to use the output from node-usb-detection to determine whether the Xkeys panel in use what (dis)connected. 

The reason for this whole exercise is to properly handle the scenario where multiple Xkeys panels are connected to a single computer, and one of them is removed. 
Potentially two XK-24's could be connected - it must then be possible to distinguish between them, and correctly handle one of them being removed and then reconnected.

nytamin, 2021-01-27:
I can definitely see that being able to handle reconnections for a previously connected panel could be useful.
If you (or anyone else) would submit a PR I'd be happy to review it.

I think we should at least add the functionality:
* Enable a consumer to listen to some event `("deviceConnected", device: HID.HID, isReconnecting: boolean)`, so that the consumer can use that to set up when a new panel is connected.
* Keep track of previously connected panels, and emit a `("reconnect")` on the panel when it's reconnected.

mountaindude, 2021-01-27:
Agree.  

Ok, I'll poke around in this and see what I can do.

nytamin, 2021-04-07:
Note to self:

I'm waiting with this a little bit until we've gotten 2.0.0 released.

I intend to add an option to `XKeysWatcher()` to enable an "automatic unitId mode", which automatically sets the unitId to something unique in order to be able to uniquely identify different panels.
It'll be an opt-in option, as it may cause unwanted behavior if the user isn't aware of what is done to the panels.

It should be able to handle most situations, with one big caveat:
If you have an xkeys-panel which has a unitId set previously, it might conflict with another panel. If this happens, two "identical" panels can be connected at the same time. We could potentially handle this by force-setting the unitId of one of the panels to something else. Some thought-work needed.

Pseudo-code:
```
on('panel-connected', (panel) => {
  if (automaticUnitIdMode) {
    if (panel.unitId == 0) { // it it is 0, we assume that it's new from the factory
      panel.setUnitId(nextUniqueForThatProductId) // the lookup-cache is stored either in memory, or preferrably on disk
    }

    const uniqueIdentifier = panel.productId + panel.unitId // the PID+UID pair is now enough to uniquely identify a panel
    if (uniqueIdentifier is in memory-cache) {
      emit('reconnected')
    } else {
      emit('new-connected')
    }
  } else {
      emit('new-connected') // default behaviour
  }
})

```

cwilling, 2021-09-04:
panel.productId + panel.unitId
may not be unique enough If a device with productId 1062 and unitId 4 is connected along with another device with productId 1065 and unitId 1, both end with uniqueIdentifier == 1066, so looks like 'reconnected' but should be a 'new-connected' since they're clearly different devices.

Rather than a simple arithmetic addition of productId and unitId, it might be better to do a string concatenation - something like:
`const uniqueIdentifier = xkeysPanel.info.productId.toString().concat('.', xkeysPanel.info.unitId);`
which would give 1062.4 and 1065.1 for the example devices.

nytamin, 2021-09-08:
I've published version 2.2.0, which adds the 'reconnect' event (when using `automaticUnitIdMode`).

@mountaindude feel free to open another issue if you have any other problems (or solutions) regarding macOS.

closed 2021-09-08 by nytamin