SuperFlyTV/xkeys #28: Support for XKE-64 JogT-bar

JeffreyDavidsz, 2021-02-10:
```
2021-1-22 17:39:31.509 Listening to device (1):
2021-1-22 17:39:31.511 manufacturer   P. I. Engineering
2021-1-22 17:39:31.519 product        XK-80 HID
2021-1-22 17:39:31.523 vendorId       1523
2021-1-22 17:39:31.524 productId      1089
2021-1-22 17:39:31.526 interface      0
2021-1-22 17:39:38.752 
2021-1-22 17:39:38.755 1 data 000040000000000000000000000a21a400000000000000000000000000000000
2021-1-22 17:39:38.833 1 data 000000000000000000000000000a21f900000000000000000000000000000000
2021-1-22 17:39:39.527 
2021-1-22 17:39:39.529 1 data 000000400000000000000000000a24a900000000000000000000000000000000
2021-1-22 17:39:39.657 1 data 000000000000000000000000000a252e00000000000000000000000000000000
2021-1-22 17:39:40.304 
2021-1-22 17:39:40.306 1 data 000000004000000000000000000a27b300000000000000000000000000000000
2021-1-22 17:39:40.433 1 data 000000000000000000000000000a282f00000000000000000000000000000000
2021-1-22 17:39:42.015 
2021-1-22 17:39:42.016 1 data 000000000040000000000000000a2e5c00000000000000000000000000000000
2021-1-22 17:39:42.096 1 data 000000000000000000000000000a2ea900000000000000000000000000000000
2021-1-22 17:39:42.607 
2021-1-22 17:39:42.609 1 data 000000000000400000000000000a30ab00000000000000000000000000000000
2021-1-22 17:39:42.697 1 data 000000000000000000000000000a310000000000000000000000000000000000
2021-1-22 17:39:43.015 
2021-1-22 17:39:43.017 1 data 000000000000200000000000000a324100000000000000000000000000000000
2021-1-22 17:39:43.120 1 data 000000000000000000000000000a32a900000000000000000000000000000000
2021-1-22 17:39:43.983 
2021-1-22 17:39:43.985 1 data 000000000020000000000000000a360500000000000000000000000000000000
2021-1-22 17:39:44.695 
2021-1-22 17:39:44.696 1 data 000000000000000000000000000a38c400000000000000000000000000000000
2021-1-22 17:39:45.351 
2021-1-22 17:39:45.352 1 data 000000000020000000000000000a3b5200000000000000000000000000000000
2021-1-22 17:39:45.895 
2021-1-22 17:39:45.897 1 data 000000000000000000000000000a3d6c00000000000000000000000000000000
2021-1-22 17:39:46.519 
2021-1-22 17:39:46.521 1 data 000000002000000000000000000a3fd900000000000000000000000000000000
2021-1-22 17:39:46.563 1 data 000000006000000000000000000a3ff800000000000000000000000000000000
2021-1-22 17:39:46.579 1 data 000000002000000000000000000a401500000000000000000000000000000000
2021-1-22 17:39:47.007 
2021-1-22 17:39:47.009 1 data 000000000000000000000000000a41c100000000000000000000000000000000
2021-2-10 09:05:15.978 Listening to device (1):
2021-2-10 09:05:15.981 manufacturer   P. I. Engineering, Inc.
2021-2-10 09:05:15.986 product        XKE-64 JogT-bar
2021-2-10 09:05:15.989 vendorId       1523
2021-2-10 09:05:15.991 productId      1325
2021-2-10 09:05:15.994 interface      0
```

JeffreyDavidsz, 2021-02-10:
What other info would you need?

nytamin, 2021-03-24:
closing this, as the XKE-64 JogT-bar is included in PR https://github.com/SuperFlyTV/xkeys/pull/32

closed 2021-03-24 by nytamin