SuperFlyTV/xkeys #33: XK-12 Joystick using wrong bits

jonwyett, 2021-03-22:
It appears the wrong bits are being queried for the joystick, at least based on my XK-12 Joystick's output. The z-axis is also being modified incorrectly. My preference is also to have (-128,-128) be bottom-left and (128,128) be top-right so I added a *-1 to the Y-axis line.

**Original Code:**
(line 109 in current release of xkeys.js via npm or 163 in the ts re-write on github)
```javascript
var d = data.readUInt32LE(7); // Joystick X
analogStates.joystick_x = (d < 128 ? d : d - 256);
d = data.readUInt32LE(8); // Joystick Y
analogStates.joystick_y = (d < 128 ? d : d - 256); 
d = data.readUInt32LE(9); // Joystick Z (twist of joystick)
analogStates.joystick_z = (d < 128 ? d : d - 256);
```

**Proposed fix:**
```javascript
var d = data[6]; // Joystick X
analogStates.joystick_x = (d < 128 ? d : d - 256);
d = data[7]; // Joystick Y
analogStates.joystick_y = (d < 128 ? d : d - 256) *-1; 
d = data[8];// Joystick Z (twist of joystick)
analogStates.joystick_z = d; //0-255 then loops
```
Please find attached 4 files:
"logHMI_XK-12Joystick_Output.txt" - Raw output from the logHMI script

When running the test code from "Getting Started" I generated these logs:
"XK-12Joystick_OrigCode.txt" - Original output using current code
"XK-12Joystick_ProposedFix.txt" - Output from same file but with my proposed modification.
"XK-12Joystick_Z-Axis.txt" - Output just showing the z-axis being rotated all the way in both directions.

I see that @michaelhatPIengineering has started collaborating on this, hopefully he can verify/test my solution with his hardware.

Cheers,
-Jonathan Wyett

[XK-12Joystick_ProposedFix.txt](https://github.com/SuperFlyTV/xkeys/files/6185030/XK-12Joystick_ProposedFix.txt)
[XK-12Joystick_Z-Axis.txt](https://github.com/SuperFlyTV/xkeys/files/6185031/XK-12Joystick_Z-Axis.txt)
[logHMI_XK-12Joystick_Output.txt](https://github.com/SuperFlyTV/xkeys/files/6185032/logHMI_XK-12Joystick_Output.txt)
[XK-12Joystick_OrigCode.txt](https://github.com/SuperFlyTV/xkeys/files/6185033/XK-12Joystick_OrigCode.txt)

michaelhatPIengineering, 2021-03-22:
Jonathan,
Thanks for keeping me honest, you are correct about the joystick. 
We are making major changes in the "Develop" branch. 

I noticed the same as you and my first fix was almost 100% the same as yours.

We then went a bit further to generalize things for our other joystick products and even multiple sticks

this.product.hasJoystick?.forEach((joystick, index) => {
				const x = data.readUInt8(joystick.joyXbyte) // Joystick X
				const y = data.readUInt8(joystick.joyYbyte) // Joystick Y
				const z = data.readUInt8(joystick.joyZbyte) // Joystick Z (twist of joystick)

				analogStates.joystick[index] = {
					x: x < 128 ? x : x - 256, // -127 to 127
					y: y < 128 ? -y : -(y - 256), // -127 to 127
					z: z, // joystick z is a continuous value that rolls over to 0 after 255
				}
			})
Take a look at xkeys.ts and products.ts in the develop branch an let us know what you think. 

Michael

jonwyett, 2021-03-22:
Ha! Well there you have it! I guess I should have checked the develop branch before posting...

I'm excited to see all the changes that are coming, particularly the inclusion of the UID, as I'm developing a library for working with multiple panels that requires knowing the Unit ID to identify them. It also has some convenience features that abstract-away some of the low-level functions in this library. For example the joystick Z-axis is returned as a delta +- value instead of the looping 0-255, things like that. I was using my own modifications to get the UID but now that it's included that's terrific! 

-Jonathan

nytamin, 2021-03-23:
The joystick delta-z value is a great idea, I'll very likely look into adding that into the upcoming version as well!
If you have any other ideas or suggestions later on , PR:s are very welcome!

michaelhatPIengineering, 2021-03-23:
Just added delta-Z to xkeytest1 example. 
It is a bit of code we usually uses to determine direction and magnitude of the z twist. 

the joystick was originally designed for a precision microscope, used to test the silicon wafer before it is cut. The wanted a knob like the traditional focus of a microscope.  
The z axis was used for the focus.
It is not return to zero and almost all others are. 
It is absolute, even during a reboot. Meaning if you put a dot on the handle the analog value will tell where that dot is ever after. 

However, the position is only 1 byte so it rolls over at 255, which makes it annoying if you just subtract the new value from the old value.  Around 0 it is a bit tricky. Look at the example where we & with 255 to solve this bump problem. 

Michael

jonwyett, 2021-03-23:
If I did submit a patch for joystick delta would you prefer it completely override the current joystick_z 0-255 value with the delta or should that be a new property like joystick_z_delta ? (or the current joystick_z could become joystick_z_abs perhaps)

michaelhatPIengineering, 2021-03-23:
The x,y,z are actual positions at any one state. So that data should be maintained as is. 
Z delta is a change of state. it is derived from the position, it could be an additional property but should not replace or hide the actual position. 
It is really position.z just like positon.x or position.y  it just seems different because it does not return to 0

jonwyett, 2021-03-23:
Yes, that was my assumption, additional property is the way to go. Thanks!

nytamin, 2021-04-07:
I've added the `deltaZ` property for the joystick event in the develop branch, which is about to be released in a few days.
You can try it out now by
```
npm install xkeys@nightly
```

Feel free to reopen this issue if there are any lingering issues.

closed 2021-04-07 by nytamin