SuperFlyTV/xkeys #34: XK-8 support

jg-k, 2021-03-26:
Hello,

Thanks for this great library!
I have an XK-8 connected to a Linux PC and get this error with node-hid not able to open it after trying to initiate it :

```
> var myXkeysPanel = new XKeys()
Uncaught TypeError: cannot open device with path /dev/hidraw2
    at new HID (/home/jg/projects/xkeys_test/node_modules/node-hid/nodehid.js:49:17)
    at new XKeys (/home/jg/projects/xkeys_test/node_modules/xkeys/dist/xkeys.js:48:28)
```
In HID.devices, it shows 4 devices from PI engineering (with a wrong product label ), all give the same error when trying to access them through nodehid.

Running as root then gives this error:
```
> var myXkeysPanel = new XKeys()
node: symbol lookup error: /home/jg/projects/xkeys_test/node_modules/node-hid/build/Release/HID_hidraw.node: undefined symbol: napi_add_env_cleanup_hook
```

any suggestions?

Thanks,

jean

nytamin, 2021-03-26:
Thanks for bringing this up.
I don't have a concrete answer for you, but I'm very interested in the solution you arrive at, to make sure that it works in this library.

Before troubleshooting too much, could you try again if it works with the [develop-branch](https://github.com/SuperFlyTV/xkeys/tree/develop)? We're currently in the works of a major version update, it might possibly work with that.

Anyway, on to troubleshooting:
I suspect that this is an issue with node-hid, so if I where to troubleshoot it, I'd start by checking that your platform [is listed there.](https://github.com/node-hid/node-hid#supported-platforms).

The next thing I'd do is try to figure something out in some boilerplate:
```javascript
// test.js:
var HID = require("node-hid")

// List the HID devices:
console.log(HID.devices())

// Pick the device to use:
var device = HID.devices()[0] // pick one here

// This should work:
var device = new HID.HID(device .path)

```
Please let me know if you've figured it out

michaelhatPIengineering, 2021-03-26:
As mentioned in issue #32 we are working on a  large overhaul. 
I am only working in windows right now to get the higher level stuff done. 

But we need input from the other OSs. 

Most of the X-keys have more than one "device" per PID, meaning, say a keyboard end point, a mouse endpoint and a consumer page where the data for this library is presented. 

Some OSs, I know Android for example, have problems with connecting to anything with a boot keyboard attached. In those cases the device must be set to a different PID, and in some cases we had to make special "Android" firmware.

As for the "Wrong product Label", that is not surprising as the base firmware in the device is the same as all 3 of the "stick" devices, it would show as a 16 key from the hardware USB string. And if it could not be read by the library it would not have been able to send a better description based on the PID in the products list. 

Can you see what PID your 8 keys is reporting?
Michael

jg-k, 2021-03-26:
both, thanks for the fast reply!

I m on ubuntu 18.04 and was testing with node 13.8, using xkeys XK-8

first error, "cannot open device" is due to OS requiring root to open device it seems.

next error:  
```
> var myXkeysPanel = new XKeys()
node: symbol lookup error: /home/jg/projects/xkeys_test/node_modules/node-hid/build/Release/HID_hidraw.node: undefined symbol: napi_add_env_cleanup_hook
```
@nytamin comment + this thread [https://github.com/hyperdivision/hid/issues/1](url) indicated some issues with this node version and node-hid (sorry have no idea how this works behind...).

running with node version 10.6.3 (picked that version from the discussion in the relevant thread) went through fine.

then next one, doing the basic test with an event listener on keypress: 

- When i press the first key once, it shows 2 key pressed, "0" and "56" and the backlight leds lights up
- When i press the second key once, it shows 2 key pressed, "8" and "40", and the 7th key lights up...
- When i press the second key once, it shows 1 key pressed, "16", and nothing else light up...
- When i press the second key once, it shows 2 key pressed, "24" and "56", and nothing else light up...
- When i repress the first key once, it shows 3 key pressed, "0", "48", "56"

sorry if i m missing obvious...

@michaelhatPIengineering thanks for the extra details :-) 
ProductID is 1130 

and noted, I ll test with development version later

cheers,
jean

Julusian, 2021-03-26:
@jg-k It sounds like you need to define some appropriate udev rules to allow applications to use the usb devices without being root.

@nytamin perhaps something in the readme for this will be a good idea. Something like:
> On linux, the udev subsystem blocks access for non-root users to the xkeys without some special configuration. Save the following to /etc/udev/rules.d/50-xkeys.rules and reload the rules with sudo udevadm control --reload-rules
```
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="05f3", MODE:="666", GROUP="plugdev"
```
I haven't tested this, but it is based on what is in the [streamdeck lib](https://github.com/julusian/node-elgato-stream-deck#linux)

nytamin, 2021-03-26:
>@jg-k Any idea why this happens?

The button-indexing is not very good at the moment.. This has been improved in the upcoming 2.0 version (to indexes 1,2,3,4 etc..) .

>@julusian perhaps something in the readme for this will be a good idea

Good point! @jg-k could you verify if this helps, and I'll add it to the docs.

michaelhatPIengineering, 2021-03-26:
The 8 stick was not correct in the first version. I suspect the time stamp may be spilling into the button bytes but not sure. And I know the led indexes were wrong. 

At this point, at least for button and led indexing we should not expend any effort to fix or understand v1

So @jg-k  if you can have a go with the develop branch, that would be the best. 

Also, the blocking of X-keys for non root users is in some OSs is a problem we can help if we add a function to change to different internal endpoint combos. (PIDs in the same device) 

Michael

michaelhatPIengineering, 2021-03-26:
@jg-k a while back we made a patch to our firmware for Linux, it would not change the issues you describe here, but since you said you are using on Linux it would be good to check. 
If you have V14 or higher you should be good. 
Else, please contact tech@piengineering.com and they will get you the tools and code to upgrade your firmware. In hardware it is not as easy to make a patch as we never know where in the supply chain you got the product.

jg-k, 2021-03-26:
@Julusian 
I added the udev rule file and reloaded the rules, but same error, cannot open device.

sorry stupid question, but could you explain to me how to run v2?...
If i clone the repo to get to the develop branch, then refer to this local package, I get: 
`Error: Cannot find module '/home/jg/projects/xkeys/dist/index.js'. Please verify that the package.json has a valid "main" entry
`

jg-k, 2021-03-26:
@michaelhatPIengineering > If you have V14 or higher you should be good.
I do have firmware V14 already, but thanks for the info :-)

nytamin, 2021-03-27:
>@jg-k could you explain to me how to run v2

Make sure you also run `npm install ` and `npm run build `, to compile the typescript into Javascript. (fyi, this is only needed when dealing with the develop/repo itself, after we've published it to npm, it'll work as usual.)

jg-k, 2021-03-27:
thks indeed, worked much better after the build...

just made a quick test and happy to let you know that V2 seems to work well with XK-8 on linux! > button press well detected and led control worked.

just the USB-detection lib didnt work on my platform, but that is a side issue/not critical and i ll get back to that later.

so now it s functioning, i m planing to connect it to my little application in a couple of weeks and will let you know of any findings after further testing.

Thanks all again for the effort put into V2! certainly a great addition to open up accessibility of X keys programming.

nytamin, 2021-03-27:
Good to hear!
Could you clarify if it worked for you _with_ Julian's udev fix, or also without?
Regarding the USB-detection, yeah let's follow that up later!

jg-k, 2021-03-27:
works without the udev fix, but i still need to run with sudo, otherwise i still get this error:

```
$ node xkey_test.js 
TypeError: cannot open device with path /dev/hidraw2
    at new HID (/home/jg/projects/xkeys/node_modules/node-hid/nodehid.js:49:17)
    at Function.<anonymous> (/home/jg/projects/xkeys/dist/xkeys.js:203:38)
```

adding the udev file and reloading the rules made no difference (i added in the udev rules folder, with the content from Julian and didn t try to look into it further or modify anything...

Julusian, 2021-03-27:
oops, I forgot to make sure the udev rules would support hidraw (not something the streamdeck lib allows for..)
Try this instead (only added the last line)

If the rules are working correctly, then you wont need to run with sudo
```
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="05f3", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="05f3", MODE="0666", GROUP="plugdev"
```

jg-k, 2021-03-29:
Thanks Julian, this last version works fine!

commit 9bc84746 (2021-04-07): chore: doc. closes #34
closed 2021-04-07 by nytamin