SuperFlyTV/xkeys #37: Closing down neatly

cwilling, 2021-05-04:
As an experiment in using xkeys, I've made myself a replica of PIEng's Device Finder application. It works fine, including multiple device re-connections,  so I'm quite pleased with it.

However when running from a terminal with `npm start` or `$(npm bin)/electron .`, closing the application window doesn't return to a prompt, rather waits until I do a Ctrl-c. I guess an event handler somewhere is still waiting for events without having been cancelled. My issue is how to close down cleanly?

If anyone is interested, the code is at [devicefinder](https://gitlab.com/chris.willing/devicefinder) where you can see (in _src/main.js_) how I'm currently trying to close things down in `app.on('window-all-closed', () => {...})` function.

Thanks,
chris

nytamin, 2021-05-04:
Thanks for reporting this bug!
It sounds to me like there must be an open handle left somewhere. There are a few tips on how to troubleshoot open handles in this discussion:
https://stackoverflow.com/questions/26057328/node-js-inspect-whats-left-in-the-event-loop-thats-preventing-the-script-from

Feel free to submit a pull request to fix this!

nytamin, 2021-05-23:
Hey @cwilling , I've had a look at the issue you've described.
I've got a fix ready, could you verify if it works for you?

In your devicefinder-repo, make the following changes:

package.json:
```
"dependencies": {
    "xkeys": "2.1.1-alpha.1
```

main.js:
```
// Line 32:
if (!win.isDestroyed()) {
	win.webContents.send("lostDevice", xkeysPanel.devicePath);
}

// Line 101:
app.on('window-all-closed', () => {
	if (process.platform !== 'darwin') {
		watcher.stop()
		.then(() => {
			ipcMain.removeAllListeners();
			app.quit()
		})
		.catch(console.error)
	}
})
```

I've tried it locally and it seems ok, please let me know if it works properly for you too.

cwilling, 2021-05-23:
After applying the changes above, I can no longer replicate the problem. Consider the fix verified!
Thanks.

closed 2021-05-24 by nytamin

nytamin, 2021-05-24:
I've published the fix to version `2.1.1`

PR #40 (2021-05-24): Fix: close devices
Closes #37