SuperFlyTV/xkeys #38: Can XKeysWatcher handle existing connections?

cwilling, 2021-05-04:
In my [Device Finder](https://gitlab.com/chris.willing/devicefinder) app., an XKeysWatcher handles connections & disconnections just fine. However, to deal with existing connections (devices already connected when the app starts), I also added `XKeys.listAllConnectedPanels().forEach(() => {...}). That seems to work but entails lots of duplicate code handling exactly the same events as for the watcher.

Is there any way for a watcher to detect existing connections? I'm guessing not but then is there a way that panels detected with listAllConnectedPanels() could be passed on to a watcher for event handling?

Thanks,
chris

nytamin, 2021-05-04:
Huh, that doesn't seem right. The XKeysWatcher _should_ pick up the initially connected devices as well (it works on my machine).

Could you provide what Node-version and which OS it doesn't work on?

Feel free to submit a pull request for fixing this, I might also have a go at troubleshooting some time during the next few days.

cwilling, 2021-05-04:
node: '12.22.1' on Linux but that wasn't related.

It was due to how I was declaring the watcher. Moving that has fixed the issue and allowed to remove listAllConnectedPanels & associated event handlers.

Thanks for making me look harder (since you said it works on your machine).

closed 2021-05-04 by cwilling