SuperFlyTV/xkeys #49: Backlight methods

cwilling, 2021-09-17:
I see the api has various methods for setting backlights - including writing to the eeprom to make changes stick. Is there any way to retrieve the currently stored settings?

nytamin, 2021-09-17:
We've only implemented methods for setting stuff at the moment. I don't know if it's possible to read the state of the backlights (@michaelhatPIengineering should know this).

Whats your use case for needing to read the states? (Myself, I always thought that setting all lights on startup would be enough.)

cwilling, 2021-09-17:
Not a particularly compelling use case -  after playing around with the backlights, I might like to go back to initial settings without having to reboot/unplug the device.

BTW like unmounting/remounting disk drives, is there an equivalent for usb devices i.e. reconnecting without replugging them? If so, could that count as a reboot and restore burnt in settings?

michaelhatPIengineering, 2021-09-17:
The very newest X-keys do have a way get the back light state. but most do not.
Our new rule is if there is a set command there should be a get also. But that has not always been the case. 

For all X-keys there is a reboot command. It essentially causes the X-keys to shut down and then the OS will re-discover the device. 

Send an output report of 238 to any X-keys and it will reboot. 

I hope to get back to adding some more output functions now that I have a node RED node (thanks Chris) I can play on my PI so that should inspire me to get back to work here

Michael

cwilling, 2021-09-17:
Further to soft reboot idea, I see there's a 0,238,0... write command available (Report 21 - described as "Reboot Device" in the data reports from PIEng). I wonder if that would work?

cwilling, 2021-09-18:
PR #50 adds rebootDevice()

cwilling, 2021-09-22:
Thanks for the merge. Since only the very newest devices can read state of backlights, rebooting a device is a pretty good solution.

closed 2021-09-22 by cwilling