SuperFlyTV/xkeys #51: Help with building

cwilling, 2021-09-18:
I made a fresh clone, installed yarn and ran it to install the dependencies - as mentioned in the README. Before making any changes at all, I ran `yarn lint`  but it gave some errors about not finding @xkeys-lib
```
chris@d8:~/src/xkeys$ yarn lint
yarn run v1.22.11
$ lerna exec --stream -- eslint . --ext .ts,.tsx
lerna notice cli v3.22.1
lerna info Executing command in 5 packages: "eslint . --ext .ts,.tsx"
xkeys-webhid: /home/chris/src/xkeys/packages/webhid/src/index.ts
xkeys-webhid:   1:15  error  "@xkeys-lib/core" is not found  node/no-missing-import
xkeys-webhid: /home/chris/src/xkeys/packages/webhid/src/methods.ts
xkeys-webhid:   1:40  error  "@xkeys-lib/core" is not found  node/no-missing-import
xkeys-webhid: /home/chris/src/xkeys/packages/webhid/src/web-hid-wrapper.ts
xkeys-webhid:   1:44  error  "@xkeys-lib/core" is not found  node/no-missing-import
xkeys-webhid: ✖ 3 problems (3 errors, 0 warnings)
lerna ERR! eslint . --ext .ts,.tsx exited 1 in 'xkeys-webhid'
lerna WARN complete Waiting for 1 child process to exit. CTRL-C to exit immediately.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```
I guess @xkeys-lib is produced when the source code is built but I'm not familiar with yarn - what's the procedure for building?

nytamin, 2021-09-18:
Try running `yarn build` from the root of of the repo, did that work? (I should probably add that to the Readme)

cwilling, 2021-09-19:
Thanks, `yarn build` works just fine.

On the other hand, `npm link` & `yarn link`, as you suggested elsewhere, aren't working as expected. One aspect of that is (maybe) that package.json has _"name": "xkeys-monorepo_  which could be confusing things. Also, probably not so important, _"version": "2.0.0"_

nytamin, 2021-09-19:
It gets a little bit complicated due to this being a mono repo, I suppose.
To set up the link for the node library of xkeys you have to
```
cd packages/node
yarn link
```
And in your test project
```
yarn link xkeys
```

Also remember to run `yarn build`every time you've made changes in the xkeys repo, before you try them out in your test project.

cwilling, 2021-09-19:
After the `yarn link xkeys` I see that test app's node_modules now has the link. However the test app's node_modules also has @xkeys-lib which is not linked. When I run the test app, it starts with Error notification saying
```
Uncaught Exception:
undefined:undefined
```
and after dismissing the notification the gui is displayed but shows no devices (whereas it normally would). When I close the gui window there's another Error notification, this time saying
```
Uncaught Exception:
TypeError: Cannot read property 'stop' of undefined
```
stop() is only used once when exiting the app and is called on the xkeysPanel object returned by the watcher's on.connected event, so looks like xkeysPanel is never properly instantiated, so undefined.

It's not really your problem so feel free to close the issue but if you have any other ideas I'm happy to try them.

nytamin, 2021-09-20:
No worries, I'd love to help!

Finally it's monday morning and I'm in front of an actual computer again. My previous attempts to help was from mobile and memory..

Here's my current attempt, which I've now tested locally. Please check it out, if it works ok I'll add it to the README.

**First**, pull the latest changes from master (I've added a convenience script in there).

**To set up the xkeys-library for linking:**
``` bash
# In the xkeys repo:
yarn lerna exec yarn link # This runs "yarn link" in all of the mono-repo packages
yarn build
```

**Set up your local test repo to used the linked xkeys libraries:**
``` bash
# In your test repo (for Node.js version):
yarn link xkeys
yarn link @xkeys-lib/core

# In your test repo (for the web-HID version):
yarn link xkeys-webhid
yarn link @xkeys-lib/core
```

**Every time after you've made any changes to the xkeys-repo:**
``` bash
# In the xkeys repo:
yarn build
```

----
After you're done, you've PR:ed your changes, they have beeen merged in and published to a new version, you probably want to unlink and used the published version again:

**To unlink the xkeys-lib from your local test repo:**
``` bash
# In your test repo (for Node.js version):
yarn unlink xkeys
yarn unlink @xkeys-lib/core
yarn --force # So that it reinstalls the regular libs
```

cwilling, 2021-09-20:
Nearly there! Before running the first `yarn lerna exec yarn link` in xkeys repo, need first to run `yarn` to install all the requirements, in particular lerna. Then, following the rest of the instructions sets up all the links.

Now the "interesting" part. When I make a change in xkeys and run `yarn build`, I can see the changes in the test app's node_modules/@xkeys-lib/core/dist/xkeys.js & xkeys.d.ts (which, of course, are now following links back to xkeys changed files). When I run the test app and invoke the new rebootDevice(), I still have an error saying the device object has no such function.

However if I do a normal installation of the test app (i.e. without links) and modify node_modules/@xkeys-lib/core/dist/xkeys.js & xkeys.d.ts by hand to be exactly the same as the linked cases, then the new rebootDevice() works as expected.

nytamin, 2021-09-20:
Huh, that's a strange problem.
What I usually do do tmwham/when having strange issues like this, is to remove the npm modules folders completely and restart from scratch

cwilling, 2021-09-22:
Yes, restarted from scratch many times ...

Anyway, the rebootDevice merge solved the immediate problem so running from a local xkeys repo is not so pressing. I'll keep trying occasionally when I have time - it would be nice to know what's not right about my setup.

BTW with updated developer instructions in the README, I think it's still good to run `yarn` by itself before running `yarn lerna exec yarn link` since that will fail if lerna is not installed already - probably a common situation for non-yarn/monorepo users.

closed 2021-09-22 by cwilling