SuperFlyTV/xkeys #55: XKeys.listAllConnectedPanels is not a function

cwilling, 2021-11-14:
I'm trying to incorporate listAllConnectedPanels to reset a local list of devices following problem of some hardware not supporting rebootDevice() correctly. However it doesn't seem to work. Here is output using CLI example on repo homepage:

```
chris@d8:~/src/xkeys-server$ node
Welcome to Node.js v14.17.6.
Type ".help" for more information.
> const { XKeys } = require('xkeys')
undefined
> XKeys.listAllConnectedPanels().forEach(() => {
... XKeys.setupXkeysPanel()
... then((xkeysPanel) => {
..... console.log(`X-keys panel of type ${xkeysPanel.info.name} listed`)
..... })
... .catch(console.log)
... })
Uncaught TypeError: XKeys.listAllConnectedPanels is not a function
```
Same result on different hardwares & node versions.

Any ideas why?

nytamin, 2021-11-14:
I haven't actually checked as I'm afk for the weekend, but could it be a simple matter of changing your import to `const XKeys = require('xkeys')`?

cwilling, 2021-11-14:
Thanks for response while you're away.
You were correct; that slight import change fixed the uncaught Syntax Error.

However then I noticed another problem with the example. It doesn't handle multiple attached devices correctly - it always lists only the first device it finds for however many devices are attached i.e. 3 devices attached => first device listed 3 times.

It seems to handle them OK if I change:
```
XKeys.listAllConnectedPanels().forEach(() => {
    XKeys.setupXkeysPanel()
    ...
```
to
```
XKeys.listAllConnectedPanels().forEach((panel) => {
    XKeys.setupXkeysPanel(panel)
    ...
```

michaelhatPIengineering, 2021-11-14:
That may help a problem I think we have with products with only 1 interface. 
The X-keys LCD can be set to PID 1321 which as only consumer in and out end points. 
I remember when we were first working on the product.ts we needed to specify the PID and the interface #. This one pid was the only one that showed as a -1. In all other cases we use the 0 interface.

michaelhatPIengineering, 2021-11-14:
got the new node working.

It is good, we may need to also make a similar for setting the PID.

I have been researching the limits of the PI4

It seems it does not like too many "devices" or endpoints in our case

I want to test the LCD with its special PID 1321. But can on seem to get it to recognize that PID.

This test on the PI4 is sort of important, because it will influence our strategy on the PI and in our future firmware.

Can you change your LCD to PID 1321 and see of you can "see" it?

MHH

cwilling, 2021-11-15:
I changed to PID 1321 but it was then no longer visible to either the Pi4 or a normal machine. Isn't that just the nature of the library though? I thought it just chose one of the possible PIDs for any device (1316 for LCD).

michaelhatPIengineering, 2021-11-15:
PID 1321 is listed as a viable PID in products.ts

XK16LCD: literal<Product>({

name: 'XK-16 LCD', // Has a 16x2 Alpha Numeric back lit LCD Display

hidDevices: [

[1316, 0],

[1317, 0],

[1318, 0],

[1319, 0],

[1320, 0],

[1321, -1],

[1322, 0],

],
It is the only PID without any extra endpoints, like a keyboard.

Because of that, it may not report any other interface so that is why we added the second entry to this list. most pids use device 0, but this one as only 1 and it seemed on my windows machine it only reported -1, l guess that means only 1 so no index number.

The reason to make this one work is that I think it may not use up the limited Attached HID devices on the PI4, one of the other PIDs may use up to 4 because of all the extra endpoints.

I have tested other PIDs in this list and they work. The only one that should not work is PID 1323, and that is not listed, it is ONLY a keyboard and therefore not a able to send any data other than keystrokes.

nytamin, 2021-11-16:
> However then I noticed another problem with the example. It doesn't handle multiple attached devices correctly - it always lists only the first device it finds for however many devices are attached i.e. 3 devices attached => first device listed 3 times.

Thanks @cwilling I noticed that there where several issues with the [example scripts](https://github.com/SuperFlyTV/xkeys/tree/master/packages/node/examples). I've updated them now to actually work.. :)

cwilling, 2021-11-16:
There's also the listAllConnectedPanels example on the front page (README.md) that needs a tweak.

nytamin, 2021-11-17:
Good point! I updated too. 👍

closed 2021-11-18 by cwilling