SuperFlyTV/xkeys #67: call a xkeys function out of the watcher

fgraux, 2022-03-01:
Hi 
thx for this new version 2.0.0.
I failed to call a setBacklight function from outside the watcher.
I have a webbased virtual keyboard that I sync to the xkeys panel: 
No issue while clicking on a physical key, I can turn both on red color.
But when I click on my virtual key, how can I set my xkeys key on red color?

nytamin, 2022-03-01:
I'm not sure what could be causing your issues.
Could you provide a code sample?

fgraux, 2022-03-01:
Sure, here is the part of my watcher where I launch tasks (websocket input & setbacklight to =xkeys) :
`const { XKeysWatcher } = require('xkeys2');
	const watcher = new XKeysWatcher();
	watcher.on('connected', (Xkeys80_1) => {
		Xkeys80_1.on('down', a => {
			//SHOW DVE
				if (a == '69' ) {
					IOSendData(['web'], 'sClick', 'dve1_box');
					if (action3 == 'cg') {
						Xkeys80_1.setBacklight(a, 'red'); //red light POWER ON selected button
						action3 = 'nocg';}
					else {
						Xkeys80_1.setBacklight(a, false); //red light POWER OFF selected button
						action3 = 'cg';}
				}
					});`

And here is the part that deals with the same command coming from the virtual key of the web app:
`		if(uiActionReceived.action == 'cg'){				//PLAY A CASPARCG TEMPLATE
			ccgChannel = uiActionReceived.channel;
			ccgLayer = uiActionReceived.layer;
			templateName = uiActionReceived.template;
			tempData = uiActionReceived.data;
			dvebox = uiActionReceived.device;
			state = uiActionReceived.state;
			divclass = uiActionReceived.divclass;
			display = uiActionReceived.display;
			file = uiActionReceived.file;
			thumb = uiActionReceived.thumb;
				if (dvebox == 'dve1_box'){ action3 = state; class3 = divclass; display3 = display; file3 = file; thumb3 = thumb;
					Xkeys80_1.setBacklight(69,'red');
					};
				
		}	`

when I click on the virtual key, Node stops and display this error: 
" Xkeys80_1.setBacklight(69,'red');                                                                                                                                                                              
                   ^                                                                                                                                                                                                                                                                                                                                                       TypeError: Cannot read properties of undefined (reading 'setBacklight')                                                                                                                                  at uiActions (D:\Caspar\LiveProd\LiveProdSrv1.js:1485:16)                                                                                                                                            at Socket.<anonymous> (D:\Caspar\LiveProd\LiveProdSrv1.js:1198:5)                                                                                                                                    at Socket.emit (node:events:520:28)                                                                                                                                                                  at D:\Caspar\LiveProd\node_modules\socket.io\dist\socket.js:435:32                                                                                                                                   at processTicksAndRejections (node:internal/process/task_queues:78:11)                                                                                                                                                                                                                                                                                                                                Node.js v17.4.0"

Have you got an idea ?
thx

nytamin, 2022-03-01:
Ok, this is likely an issue of scope. The Xkeys80_1-variable you use in your second example is not the same as the one in the first example.

You could dock something in the spirit of 
```
var myGlobalXkeys
watcher.on('connected',` (Xkeys80_1) => {
  myGlobalXkeys = Xkeys80_1
} 
```
And then use myGlobalXkeys in your second example.

This will of course cause issues if you connect another panel (or disconnect the first one), bit I'll leave that to you to figure out 🙂

fgraux, 2022-03-01:
Simply but works Great
thx

closed 2022-03-01 by nytamin