SuperFlyTV/xkeys #92: Intermittent "NotAllowedError: Failed to write the report." on Chrome v117+

KhangxLi, 2023-10-24:
The following uncaught exception (or promise rejection, not sure) `"NotAllowedError: Failed to write the report."` gets thrown from within function `setupXkeysPanel`.

On Chrome versions 117+ (tested on 117, 118, 120), this happens consistently until Chrome is restarted* ( can use chrome://restart ). *The issue gets resolved only after a random number of Chrome restarts and sometimes requiring XKeys to be replugged. The issue can then be reintroduced after a certain random number of Chrome restarts again.

The issue is never reproduced on Chrome v115 and below. Chrome v116 behaviour is untested.

In Chrome device log ( chrome://device-log/ ) I see either one of these 2 errors when the issue occurs when calling setupXkeysPanel:
- `Output report buffer too long (36 > 2).`
- `This device does not support output reports.`

Reproduced with:
- XKeys: vendorId=1523, productId=1089, name='XK-80 HID'

Any ideas?

michaelhatPIengineering, 2023-10-24:
This is the first we have heard of. The output report has stayed the same. 
It may be trying to write to the wrong end point. 
The X-keys  is several "devices" to the OS as it is a compound USB device. 
Be sure to check the write length is longer than 20 or HID usage page 0xc  (12 dec
)

Writing to device "0" for example does not always work if the OS enumerates them in a different order.

Julusian, 2023-10-25:
This was pretty easy to reproduce, and I could see that when approving the device it returns 3 different objects where we were always taking the first of this list.

Filtering this to only return and accept ones with usagePage 12 has made the issue go away for me

KhangxLi, 2023-10-25:
Correction: Issue might have been present on earlier Chrome version (115) and was exposed by a combination of Windows upgrade and Chrome update.

michaelhatPIengineering, 2023-10-25:
I reviewed with our firmware engineer; she recommends:
filter only on:
VID
PID
Usage page 12
write length > 20
to get the "device" supported by this lib. 
other things are redundant or not consistent over all OSs

KhangxLi, 2023-10-25:
thanks guys, I'm able to fix it locally since I can access these fields. 

Will update my package once https://github.com/SuperFlyTV/xkeys/pull/93 is merged since it seems to also take care of an uncaught promise rejection.

Won't close this issue myself until above is merged

closed 2023-11-02 by nytamin

nytamin, 2023-11-02:
This is now fixed in version `v3.0.1`, thanks for contributing!

michaelhatPIengineering, 2023-11-27:
Johan,

I see Peter is being persistent.

Just because VEC uses the PI VID does not mean the product is an X-keys or even a PI product. It is not.

We design a lot of products for other companies, and sometimes use our VID if they do not have one. But we are usually under NDA and other contractual obligations not to release software for their products. Not can we support a product that we do not fully control the production and sales.

It is not in PI's or VEC's business model that their product be considered an X-keys product.

I think some people think Open Source means the right to hack anything they like without consideration to the ramifications to the people they are hacking.

It would set a bad precedent for me and my other customers if we published the details of their products.

The basic issue is that I do not make any revenue on the VEC foot pedal.  It was designed specifically for their software customers in transcription to have a dedicated hardware for their product. As such the cost structure is very different.

Since the VEC product is made in China we have very little control on how or where is it sold.

He found an interesting loop hole with the Companion leading him back to your site.

In fact this will lead a lot of products that we do not make any more or made for other people back to your lib.

So perhaps we should put:

if (PID == PI_Engineering_PID) {
  if (VID == 210 or > 1023) {
   do something with xkeys library
  } else {
   go away...
  }
}

Again, it is your repo so you may do as you like but VEC is not X-keys

Thanks for your understanding.

Michael

PR #93 (2023-11-02): fix: filter for correct usage/usagePage when finding xkeys devices
fixes #92

The filtering might be overly strict, and needs to be verified on windows and macos to ensure the required properties are defined.

Tested with a XK24 and XK12-Jog