SuperFlyTV/xkeys #111: [question] Why are WebHID calls to `sendReport` queued with concurrency 1?

loucadufault, 2024-11-29:
I am looking to perform multiple writes to the device in a synchronous manner. The current wrapper method implementation sequentially queues the `sendReport` promises:

https://github.com/SuperFlyTV/xkeys/blob/e9d925e22d184775e252c1991a1105bfedc14311/packages/webhid/src/web-hid-wrapper.ts#L24-L32

Is there a reason this is done? Wondering if there were any observed issues or concerns with firing for example a few dozens `sendReport` calls without waiting for any of them to resolve (before firing off the next one).

The context is I am looking to perform some cleanup of the device on [`beforeunload`](https://developer.mozilla.org/en-US/docs/Web/API/Window/beforeunload_event), which must be sync.

Julusian, 2024-12-02:
I think this is a little related to https://github.com/SuperFlyTV/xkeys/issues/106

I suspect what happened is this code was based on the streamdeck library, which uses a queue because it sends messages in multiple packets that need to not be interleaved with other calls to `write` https://github.com/Julusian/node-elgato-stream-deck/blob/6240d8853272a4f651b835f332ef14f77c512ee5/packages/webhid/src/hid-device.ts#L45-L51 (Added in commit `fix: ensure order of hid.sendReport calls when image is multiple packets. concurrent draws previously would result in some corruption`)

For the xkeys where the writes are only ever a single hid packet, it does look unnecessary

loucadufault, 2024-12-03:
> However, if you're looking to send something in a synchronous manner, I don't think you'll be able to do that anyway, since the webhidDevice.sendReport() is asynchronous in itself, so you'll have no way of knowing if the commands have actually been sent, right?

That's acceptable for my use case, it's a fire and forget as the app is closing.

closed 2024-12-05 by nytamin