SuperFlyTV/xkeys #119: Support for KVM version?

loucadufault, 2025-07-30:
The [XKE-128 KVM keyboard](https://xkeys.com/kvm128.html) does not seem to be supported by this library for Web HID (not returned from permissions request).

Is there any reason why, or rather any limitation that would prevent this keyboard from being supported the same as the regular XKE-128 model?

I can see they have different usage page number for the `collections[0]`; regular model is 002, whereas KVM model is 006.

michaelhatPIengineering, 2025-07-30:
The XKE-128 KVM keyboard is a special version designed to work with KVM systems that do not like any endpoints other than a keyboard end point. It is designed to be setup with P.I. Engineering Inc. software to send USB HID keycodes only.  This lib is intended to use X-keys with their raw consumer page HID report.  The 006 usage page is the keyboard page. 

The hardware can be set into is "programing mode" where the raw data is presented on the consumer endpoint just the same as the regular 128 model. Then it is possible to make it stay in that mode with a special op code (I do not recall the exact one) it is in the SDK

But for the purposes of this lib the KVM devices are not natively anything other than a keyboard. So it would be difficult to support them.

Julusian, 2025-07-30:
To add to that, most oses don't allow applications to open keyboard hid endpoints for security reasons

loucadufault, 2025-07-30:
@michaelhatPIengineering 

Interesting, I did see this in the documentation:

> To convert the device to PID #1 when it is currently in PID #2 unplug the device then replug it in and press the Scroll Lock key repeatingly 10-15 times until the red LED flashes. The device gives 10 seconds after reboot to detect the Scroll Lock toggles. If it is not done quick enough, hotplug and try again. This procedure requires a regular keyboard to be attached to the computer for the Scroll Lock key. The device will revert to PID #2 (KVM mode) on reboot unless Reboot Mode (see output reports) was changed.

And then:

> PID #1 Endpoints: Endpoints: Consumer Usage Page Input and Output (Hid Usage Page 12, Hid Usage 1), Boot Mouse (Hid Usage Page 1, Hid Usage 2), Boot Keyboard (Hid Usage Page 1, Hid Usage 6).

Which looks identical to the regular XKE-128 specification:

> PID #4 Endpoints: Consumer Usage Page Input and Output (Hid Usage Page 12, Hid Usage 1), Boot Mouse (Hid Usage Page 1, Hid Usage 2), Boot Keyboard (Hid Usage Page 1, Hid Usage 6).

So assuming I manage to get it and have it stay in PID \#1, I can expect it to behave the same as the regular XKE-128?

Also, do you have any additional info on what a reboot means in this context? Is it just the power cycling that occurs when it is hotplugged?

michaelhatPIengineering, 2025-07-30:
it should have the same report as the regular 128.   We mean repowering, it will revert to the KVM mode.

Julusian, 2025-07-30:
From the chrome docs and chromium code:
https://developer.chrome.com/docs/capabilities/hid
https://source.chromium.org/chromium/chromium/src/+/main:services/device/public/cpp/hid/hid_report_utils.cc;l=28-35

I dont know the xkeys reports well enough to say for certain that it will get blocked by that block, but it looks like it could.